adafruit/Adafruit_AM2315 #8: Am2315 read Temp/Humidity

Giang2211, 2016-10-28:
Hi everyone,

I try to read data from AM2315 that i followed below link:

Everything and harware setup are Ok ! (I can detected address of AM2315). However, I cannot red out data as attached picture
http://sopwith.ismellsmoke.net/wp-content/uploads/2014/03/PI-How-To-AM2315.pdf

SSH command line interfaces:
>>> from AM2315 import*   //OK
>>>sensor=AM2315()         //OK
>>> sensor.temperature()   // NOT OK

[image]

rjgates, 2016-11-23:
#Hacked Premium account giveway on telegram: https://telegram.me/G3oPx

ladyada, 2019-06-16:
this is a different library you're talking about.

closed 2019-06-16 by ladyada