adafruit/Adafruit_AM2315 #14: Lib does not work with ESP8266

bonjour81, 2019-04-17:
Hello,

The lib does not works with ESP8266 (tested on wemos D1 mini pro, but works fine on a huzzah32)

While `am2315.begin()` return true as expected, temp & humi are always 'nan'.

The modified library https://github.com/switchdoclabs/SDL_ESP8266_AM2315 also based on adafruit library works, but I was not able to establish the exact cause of the issue.

(take care this SDL lib has currently no handling of negative temps, I opened issue #2 for this).

ladyada, 2019-06-16:
just tested latest with esp, works fine

kintel, 2022-02-05:
I ran into exactly this problem on an esp32. After testing the switchdoclabs fork to confirm that it works, I read the code and datasheet and I suspect that the wakeup timing isn't working on all platforms.

If I modify the wakeup code to the following, this library starts working again:
```
  // Step one: Wake-up sensor
  // 10 * 8 clock cycles @ 100kHz ~= 800us
  i2c_dev->write(reply, 10);
  i2c_dev->write(reply, 1);
```

It's a bit odd that just sending a larger packet doesn't work (I've tried sending 20 dummy bytes but it doesn't help).
I haven't done any low-level oscilloscope debugging of this (yet)..

kintel, 2022-02-06:
Confirmed on an oscilloscope that the wakeup timing doesn't work as expected:
The intent of `i2c_dev->write(reply, 10)` is to clock out 10 zero bytes to allow the sensor to wake up. However, since the AM2315 sensor doesn't actually ACK the address byte when it's sleeping, the Arduino I2C implementation will actually return upon receiving the first NACK, and only clock out a single byte (the address).

On classic Arduinos this seems to work, possibly because they are slow enough to allow the sensor to wake up naturally before sending the next I2C command. However, on faster microcontrollers or controllers with different underlying I2C implementations, this won't work and another technique is needed to allow the sensor to wake up.

kintel, 2022-02-07:
FYI: While looking into this, I found and tested https://github.com/RobTillaart/AM2315. I haven't yet looked into exactly how the communication is implemented in that lib though.

I also found this article which talks about how they got the sensor working on an ESP8266: https://www.switchdoc.com/2016/04/am2315-highly-reliable-esp8266-arduino-driver/

caternuson, 2022-02-08:
Seems to be ESP32 specific.

Connected AM2315 to Feather ESP8266:
[image]

Running example sketch from library:
https://github.com/adafruit/Adafruit_AM2315/blob/master/examples/am2315test/am2315test.ino

And getting generally expected output:
[image]

However, same setup with a Feather ESP32 produces (scrolls forever):
[image]

Using the wake up procedure mentioned in the article linked above produces:
[image]

It's working, but requires an extra second, on top of the current 2 second delay, before it can be read again. So, as also mentioned in same article, read rate appears to be once per 3 seconds.

Will PR some changes.

caternuson, 2022-02-08:
@kintel If possible, can you please test PR #24 on your setup. I'm wondering if the delay needs to be bumped up to 3 seconds from the current 2 seconds. I saw no issues with the current 2 secs so PR only has update for wake up.

kintel, 2022-02-08:
I tested your PR branch and it works! Left a minor comment on your PR.

Btw., as an extra datapoint, the original library worked out of the box on an ESP32-S2. These devices have binary blob HAL libraries so I'm not too motivated to dig too far into that : /

caternuson, 2022-02-08:
Please try the 2.2.1 release of the library when it becomes available:
https://github.com/adafruit/Adafruit_AM2315/releases/tag/2.2.1

caternuson, 2022-02-11:
Closing. Seems like the release above fixed this.

closed 2022-02-11 by caternuson