adafruit/Adafruit_AM2315 #15: Add support for reading device information

ScholliYT, 2019-11-18:
I would love to see support for reading device information with this library. 

As the [datasheet ](https://cdn-shop.adafruit.com/datasheets/AM2315.pdf) (page 17) says there're registers for **model, version and ID**.

caternuson, 2019-11-21:
What is the use case for this? Is this something you think you could PR?

ScholliYT, 2019-11-21:
Yeah I'm working on a PR for that. My use case might be something specific, as I got multiple sensors and want to identify them by their hardware fingerprint. 

Don't know if this features fit well into the library, as it's really simple and straith forward currently.

ladyada, 2019-11-24:
thanks for the PR! let us know when its ready to review

ScholliYT, 2019-12-10:
I had no time to test this yet. Maybe next weekend!

caternuson, 2021-09-08:
Closing. Looks like this sensor does not support this, at least per available documentation. 

See discussion in #16

closed 2021-09-08 by caternuson