adafruit/Adafruit_AM2315 #18: Unnecessary check

me21, 2020-06-05:
There is unnecessary check at
https://github.com/adafruit/Adafruit_AM2315/blob/030f7bf998bc67946e3944eca98d6e797ba13014/Adafruit_AM2315.cpp#L58

According to C++ unsigned integers arithmetic, if `millis()` wraps around, the subtraction `millis() - lastreading` would still give the correct result.

caternuson, 2021-09-08:
Maybe unnecessary, but doesn't hurt to have it and be more explicit and not get caught by any implementation dependent behavior.

closed 2021-09-08 by caternuson

me21, 2021-09-09:
There cannot be any implementation defined behavior, as unsigned integers arithmetic is defined in C++ standard. Moreover, there are two calls to millis() whereas one call with storing the result would be sufficient.
Though I agree these optimizations are minor and probably do not influence execution time much.