adafruit/Adafruit_AM2315 #20: Example code not compiling for Arduino Nano 33 BLE.

JamesCarterYoung, 2021-04-04:
- Arduino board:  Arduino Nano 33 BLE

- Arduino IDE version (found in Arduino -> About Arduino menu):  Arduino 1.8.13 (Windows store 1.8.42.0)

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):  

Attempting to compile the example code for the AM2315 spits out an "Error compiling for board Arduino Nano 33 BLE." Below is the code, and below that are the error messages.

```
#include <Wire.h>
#include <Adafruit_AM2315.h>

/*************************************************** 
  This is an example for the AM2315 Humidity + Temp sensor

  Designed specifically to work with the Adafruit AM2315 Sensor 
  ----> https://www.adafruit.com/products/1293

  These displays use I2C to communicate, 2 pins are required to  
  interface
  Adafruit invests time and resources providing this open source code, 
  please support Adafruit and open-source hardware by purchasing 
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.  
  BSD license, all text above must be included in any redistribution
 ****************************************************/

// Connect RED of the AM2315 sensor to 5.0V
// Connect BLACK to Ground
// Connect WHITE to i2c clock
// Connect YELLOW to i2c data

Adafruit_AM2315 am2315;

void setup() {
  Serial.begin(9600);
  while (!Serial) {
    delay(10);
  }
  Serial.println("AM2315 Test!");

  if (! am2315.begin()) {
     Serial.println("Sensor not found, check wiring & pullups!");
     while (1);
  }
}

void loop() {
  float temperature, humidity;

  if (! am2315.readTemperatureAndHumidity(&temperature, &humidity)) {
    Serial.println("Failed to read data from AM2315");
    return;
  }
  Serial.print("Temp *C: "); Serial.println(temperature);
  Serial.print("Hum %: "); Serial.println(humidity);

  delay(2000);
}
```

```
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Nano 33 BLE"

C:\ ... \Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.2\libraries\Wire\Wire.cpp: In member function 'void arduino::MbedI2C::receiveThd()':

C:\ ... \Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.2\libraries\Wire\Wire.cpp:145:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

      if (c <= sizeof(buf)) {

          ~~^~~~~~~~~~

C:\ ... \Documents\Arduino\libraries\Adafruit_AM2315\Adafruit_AM2315.cpp: In member function 'boolean Adafruit_AM2315::readData()':

C:\ ... \Documents\Arduino\libraries\Adafruit_AM2315\Adafruit_AM2315.cpp:79:19: error: call of overloaded 'write(int)' is ambiguous

   _i2c->write(0x00); // start at address 0x0

                   ^

In file included from C:\ ... \Documents\Arduino\libraries\Adafruit_AM2315\Adafruit_AM2315.h:22:0,

                 from C:\ ... \Documents\Arduino\libraries\Adafruit_AM2315\Adafruit_AM2315.cpp:17:

C:\ ... \Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.2\libraries\Wire/Wire.h:52:20: note: candidate: virtual size_t arduino::MbedI2C::write(uint8_t)

     virtual size_t write(uint8_t data);

                    ^~~~~

In file included from C:\ ... \Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.2\cores\arduino/api/ArduinoAPI.h:31:0,

                 from C:\ ... \Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.2\cores\arduino/Arduino.h:51,

                 from C:\ ... \Documents\Arduino\libraries\Adafruit_AM2315\Adafruit_AM2315.h:18,

                 from C:\ ... \Documents\Arduino\libraries\Adafruit_AM2315\Adafruit_AM2315.cpp:17:

C:\ ... \Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.2\cores\arduino/api/Print.h:50:12: note: candidate: size_t arduino::Print::write(const char*)

     size_t write(const char *str) {

            ^~~~~

exit status 1

Error compiling for board Arduino Nano 33 BLE.
```

Seems to me that the write() function in the AM2315 library is ambiguous with the write function in the mbed OS. Is this an error in the Arduino itself, or can I fix it somehow?

ladyada, 2021-04-04:
@per1234 any suggestions for where to fix?

per1234, 2021-04-06:
Hello. The fix for this issue was made in the Arduino Mbed OS Boards platform just last week:
https://github.com/arduino/ArduinoCore-mbed/pull/171/commits/2958bc00a985cbe5b63400fb4afa117453750255

Of course, that won't reach the users until the next release of that platform. So if you want to fix it right away here, you could add the equivalent cast to resolve the ambiguity:
```diff
index e5fecec..5ed3705 100644
--- a/Adafruit_AM2315.cpp
+++ b/Adafruit_AM2315.cpp
@@ -76,7 +76,7 @@ boolean Adafruit_AM2315::readData(void) {
   // OK lets ready!
   _i2c->beginTransmission(AM2315_I2CADDR);
   _i2c->write(AM2315_READREG);
-  _i2c->write(0x00); // start at address 0x0
+  _i2c->write((uint8_t)0x00); // start at address 0x0
   _i2c->write(4);    // request 4 bytes data
   _i2c->endTransmission();
```

JamesCarterYoung, 2021-04-07:
What a legend. Thank you so much.

closed 2021-04-07 by JamesCarterYoung